cnlohr/nosdk8266 #1: What does it mean to compile without an SDK?

leopck, 2016-10-08:
Sorry for my ignorant question but I just checked out your project here and inside the title description it says `Trying to make ESP8266 projects without any SDK whatsoever.`

So I was wondering what does it mean to compile a project without any SDK? Does it mean that the project is smaller in memory size or increase in performance of GPIO toggling since it doesn't have to go thru the SDK?

cnlohr, 2016-10-08:
There is sadly no improvement in GPIO performance over direct GPIO access without SDK (assuming you use the registers rather than the GPIO functions).

The really big improvement here is size.  I have an implementation of espusb down to 2.5kB!

leopck, 2016-10-08:
HOLY SHIT! Congrats man! That's awesome! What about WiFi? Is WiFi supported already?

cnlohr, 2016-10-08:
No.  There is zero intent to support wifi.  Wifi (And the network accoutrements) are what suck up all the space.  But, even if I wanted to, it would be exceedingly difficult as most of it is locked away, hidden behind binary blobs.

leopck, 2016-10-08:
Oh, that's very unfortunate :( but it can't be helped if the Wi-Fi is the one that takes all the space. So this project is tasked more for applications that requires memory optimization and no internet connectivity? Is this project going to integrate with esp82xx? Is it even possible considering that esp82xx is based on the espressif SDK?

cnlohr, 2016-10-08:
Sort of.  I made changes to esp82xx to make it "compatible" with this project.  The real purpose is so espusb can be run without an SDK so it can become a bootloader.  The "bootloader" folder in espusb currently implements this.  I just figured I'd leave this here for anyone else who has particularly demanding projects.

leopck, 2016-10-08:
That's really neat :) Especially for memory constraint applications. I believe this project has many applications to it. I'm curious, since there's no WiFi and the memory size is reduced without the SDK dependency.... Does it also mean that the power consumption has been reduced?

cnlohr, 2016-10-08:
Not from what I've seen.  Looks like it's about the same as a normal wifi sleep.  Processor is still screaming at 80 MHz.  I don't know how to put it to sleep.

leopck, 2016-10-08:
Alright thanks for your reply. I think I'll close this since the main question has already been answered :D

closed 2016-10-08 by leopck