cnlohr/nosdk8266 #2: Fatal exception (29):epc1=0x40001635, epc2=0x00000000, epc3=0x00000000, excvaddr=0x13233020, depc=0x00000000

leopck, 2016-10-08:
Hi, I compiled the program via `make && make burn` and I got these errors that keeps repeating the whole time. Am I missing some steps?

```
Fatal exception (29): 
epc1=0x40001635, epc2=0x00000000, epc3=0x00000000, excvaddr=0x13233020, depc=0x00000000
Fatal exception (29): 
epc1=0x40001635, epc2=0x00000000, epc3=0x00000000, excvaddr=0x13233020, depc=0x00000000
Fatal exception (29): 
epc1=0x40001635, epc2=0x00000000, epc3=0x00000000, excvaddr=0x13233020, depc=0x00000000
```

cnlohr, 2016-10-08:
I don't know.  I did see that quite a bit whenever I got anything wrong.  Pulling from this tree, no changes?

leopck, 2016-10-08:
Erm, pulling from this tree meaning pulling from the master branch right? Yes, I pulled the latest one at that moment and I just checked my git is still the latest. I'm not sure what's causing the error, it's crashing before I can do anything. Again, I'm still using my NodeMCU, I've made sure that the flash_mode = dio this time haha.

cnlohr, 2016-10-08:
Flash mode should be able to be either without issue.  I am using an ESP-12-F.  Hmm, mind putting a printf(...) among the various boot parts --- also, don't call the code that sets the crystal.  The baud rate will be 75k, but may give you insight into the boot process.

MrAda, 2017-01-09:
Any progress on that NODEMCU?

cnlohr, 2017-01-09:
I don't think I understand what the problem is specifically here?

MrAda, 2017-01-09:
I should have pointed my question to leopck.  I have a NODEMCU as well and was curious if he was successful.

cnlohr, 2017-01-09:
Aah, then make sure to poke @leopck  :-p

leopck, 2017-01-10:
@cnlohr Woops! I'm sorry guys! I forgot to update this issue! Working on several projects at the same time makes you forget things -_-|||

@MrAda Yes after cnlohr updated a few things a few builds ago, this issue disappeared 'magically'. I have no idea what caused this issue at all since I read thru the changes done, and it didn't relate to this Fatal exception. But after I deleted the old folder and cloned this new build, this fatal exception no longer shows up 👍 

You might need to perform `sudo apt install bc` before that.

MrAda, 2017-01-10:
@leopck    I need to have WIFI so I cannot use @cnlohr 's branch.  Since you and I both have NODEMCU's I am curious if you still use the LUA firmware?  I have have found it very buggy.

leopck, 2017-01-11:
@MrAda I think what you're looking at here isn't what you want. This project "nosdk8266" is meant for a different purpose (mainly bootloader). You might want to check out cnlohr's other projects which enables you to have wifi and yet program your own esp for other purposes. 
[https://github.com/cnlohr/esp82xx](https://github.com/cnlohr/esp82xx)

Regarding your question, I have stopped using LUA firmware due to the RAM issue which at that time couldn't be resolved. The issue was that if you tried running your esp using LUA for a few weeks then the esp would crash due to lack of RAM. And yes, I do feel LUA is buggy as well.

If you're interested in developing on nodemcu which is esp8266, you can just use something like [Arduino for ESP8266](https://github.com/esp8266/Arduino) or [esp-open-sdk](https://github.com/pfalcon/esp-open-sdk) or [esp82xx](https://github.com/cnlohr/esp82xx)

Since this topic isn't related to the issue that I created above I think I can close this issue.

closed 2017-01-11 by leopck

MrAda, 2017-01-11:
@leopck I do remember a person complaining about the RAM issues.  I was having similar issues.  Thanks for those links.  I'll check them out.  You can close this topic! :-)