cnlohr/nosdk8266 #5: Use this with gdb?

SmallRoomLabs, 2017-02-10:
I'm using the nosdk (thank you for this!) as a basis for a project where I have a bigger need for a 64K large array in ram than I have for wifi.  But I'd also love to be able to use the gdb debugger as well to track down some bugs I've got.

I gave the [esp-gdbstub](https://github.com/espressif/esp-gdbstub) project a quick go together with the nosdk, but no joy...   I'm too much of a rookie when it comes to the finer points of ESP coding and the SDK in general to assess this problem myself without spending more time on getting gdb possibly working than it would take me to fix the bug without the assistance of gdb,

But just to check here first before giving up - is it possible to easily add the gdbstub stuffs to nosdk?  If so it would be great to have it in the Makefile as an option already.  Hint-hint nudge-nudge ;-)

cnlohr, 2017-02-13:
Oh man... I just don't know the first thing about how GDB works.   I am a printf debugger, myself :(.  I can understand why you would want it for something as sophisticated as the ESP.  I just never did.  I don't even know who to "at" here for help with this :( Maybe post on the esp8266 forums?

SmallRoomLabs, 2017-02-13:
Ollrighty.... I solved my current issue by stripping down the code ( that originally was working fond under the noos SDK, but just crashed when moved onto the nosdk framework) down to its bare minimum to isolate the lines related to the point of crash.

It turned out that I missed copying the call for a initializer when I moved to codebase to nosdk. ;-(

But I'll give the gdbstub thingie a go on some noos projects to get up to speed with using it and update this issue and maybe do a pull request if I ever get it working on  nosdk.

closed 2017-02-13 by SmallRoomLabs

cnlohr, 2017-02-13:
btw you can give esp82xx's linker script and this SDK a shot -> https://github.com/cnlohr/esp_nonos_sdk

cnlohr, 2017-02-13:
err... those mostly focus on lower iram usage, though :(