cnlohr/nosdk8266 #8: Compiling esp-open-sdk

GanesanGuru, 2018-03-16:
Hi fellas,

I just now started working with esp-open-sdk. All I've doubts are basic one's, yet they are vital. I cloned the git and I've ran the "make" command. It generated four files.
*sdk - soft link for ESP_SDK generated
*ESP_SDK itself
*xtensa-lx106-elf
*user_rf_cal_sector_set.o

What is xtensa-lx106-elf? Is it compiler or a toolchain? Because esp-open-sdk/xtensa-lx106-elf/bin has all in it (compiler,assembler,linker,debugger). SO HAS TO BE A TOOLCHAIN??

Similarly explain what are crosstool-NG,lx106-hal,esp-open-lwip which are downloaded from git hub.

PS:I'm newbie to these knid of things and all. So please suggest me what are the things that I should go through to understand these.

cnlohr, 2018-03-16:
You should only need the esp-open-sdk toolset installed to your home folder.

It _should_ have everything to build this project.

What happens when you install it and you type "make" in this project?

GanesanGuru, 2018-03-17:
Hi...

Thanks for reaching me. When I installed esp-open-sdk, it is worked fine. The make is successfull and I've added the path to $PATH variable in ~/.profile. When I tried to with the example blinky, the make is failed stating, _blinky.c:1:21: fatal error: ets_sys.h: No such file or directory_

**Directory** is /home/esp8266/Desktop/ESP_OPEN_SDK/esp-open-sdk/examples/blinky

**The ~/.profile contents are............................**

# ~/.profile: executed by the command interpreter for login shells.
# This file is not read by bash(1), if ~/.bash_profile or ~/.bash_login
# exists.
# see /usr/share/doc/bash/examples/startup-files for examples.
# the files are located in the bash-doc package.

# the default umask is set in /etc/profile; for setting the umask
# for ssh logins, install and configure the libpam-umask package.
#umask 022

# if running bash
if [ -n "$BASH_VERSION" ]; then
    # include .bashrc if it exists
    if [ -f "$HOME/.bashrc" ]; then
	. "$HOME/.bashrc"
    fi
fi

# set PATH so it includes user's private bin if it exists
if [ -d "$HOME/bin" ] ; then
    PATH="$HOME/bin:$PATH"
fi
export PATH=/home/esp8266/Desktop/ESP_OPEN_SDK/esp-open-sdk/xtensa-lx106-elf/bin:$PATH

PATH=$PATH:/opt/esp-open-sdk/esptool

**The contents of Makefile in example/blinky is as follows**

CC = xtensa-lx106-elf-gcc
CFLAGS = -I. -mlongcalls
LDLIBS = -nostdlib -Wl,--start-group -lmain -lnet80211 -lwpa -llwip -lpp -lphy -lc -Wl,--end-group -lgcc
LDFLAGS = -Teagle.app.v6.ld

XTENSA_TOOLS_ROOT ?=/home/esp8266/Desktop/ESP_OPEN_SDK/esp-open-sdk/xtensa-lx106-elf/bin

SDK_BASE ?=/home/esp8266/Desktop/ESP_OPEN_SDK/esp-open-sdk/sdk/

blinky-0x00000.bin: blinky
	esptool.py elf2image $^

blinky: blinky.o

blinky.o: blinky.c

flash: blinky-0x00000.bin
	esptool.py write_flash 0 blinky-0x00000.bin 0x10000 blinky-0x10000.bin

clean:
	rm -f blinky blinky.o blinky-0x00000.bin blinky-0x10000.bin

cnlohr, 2018-03-19:
can you send me output from make?

cnlohr, 2018-03-19:
Like type make and report output.

GanesanGuru, 2018-03-19:
After I entered make:

xtensa-lx106-elf-gcc -I. -mlongcalls   -c -o blinky.o blinky.c
blinky.c:1:21: fatal error: ets_sys.h: No such file or directory
 #include "ets_sys.h"
                     ^
compilation terminated.
make: *** [blinky.o] Error 1

cnlohr, 2018-03-19:
Your compile line should look more like:

~/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc  -DPICO66 -flto -DPICONOPRINT -DUSE_I2S -Os -Iinclude -nostdlib  -DMAIN_MHZ=320   -mno-serialize-volatile -mlongcalls -g  pico.c src/nosdki2s.c src/startup.S src/nosdk8266.c  -T ld/linkerscript.ld -T ld/addresses.ld -o image.elf

Or something like that.  Also, I don't know what blinky you're looking for.

GanesanGuru, 2018-03-20:
I'm unable to follow your compile line. I've followed this video tutorial
https://www.youtube.com/watch?v=T-oSjMCmNYk

In that check after 12.45 minutes. He used to build the example blinky program.

The esp-open-sdk has example in it and that directory has an blinky program. I'm executing that program only. The file location is given as,

/home/esp8266/Desktop/ESP_OPEN_SDK/esp-open-sdk/examples/blinky

cnlohr, 2018-03-20:
This project has nothing to do with that project or any of the related projects in esp-open-sdk.  It only uses the esp-open-sdk compiler.

This project is only for using the ESP8266 is a very unusual way, without any SDK or wifi at all.

GanesanGuru, 2018-03-23:
Yes I know that fact. I'm just curious to find out what I've missed. As I'm new to these kind of things I'm not sure where to find the answer. That's all.

cnlohr, 2018-03-23:
I'm really not the right person to help you there, since I've never worked with that sort of stuff... try reddit /r/esp8266 or another forum?  Sorry :-/

closed 2018-03-23 by cnlohr