cnlohr/nosdk8266 #9: Get time functions

lealgo, 2018-10-15:
Hi,

Great project! Working just fine on my NodeMCU. I'm trying to get the current time in ms, something like the Arduino millis() function. I've tried the macro NOW() but it always returns 1. What is the proper way to do it?

Best regards,
Leandro

cnlohr, 2018-10-15:
It's a little tricky, but you need to use CCOUNT.   For a little more details you can see this post here: https://github.com/esp8266/Arduino/issues/182

But, to copy-and-paste it... 

```#define RSR_CCOUNT(r)     __asm__ __volatile__("rsr %0,ccount":"=a" (r))
static inline uint32_t get_ccount(void)
{
    uint32_t ccount;
    RSR_CCOUNT(ccount);
    return ccount;
}```

The 'rsr' assembly instruction takes only maybe 2-3?? (not exactly sure) cycles to complete, so anywhere you really need high performance, you can use just the ASM volatile, but, for a regular function call you can use the rest of it.

Thankfully, you get CCOUNT with the nosdk.

cnlohr, 2018-10-15:
OOH! FORGOT TO EXPLAIN:  This function returns cycle count, it is a number that just counts upwards, continuously.  And it counts at the core CPU... So, you may need to put a little more logic in if you want "the number of milliseconds since system start.  This can be a tad of a problem because ccount is a 32-bit counter, so at 160 MHz, it cycles every 26 seconds... You'll need to keep track of the total elapsed time some other way.

lealgo, 2018-10-15:
Hi:

It works like a charm. thank you! Maybe I could use it for measuring small time intervals... By the way, I was checking out SDKnoWiFi and saw some time functions implemented like system_get_rtc_time() and system_get_time(); that could do it for me. Only that I can't get that sdk to work on my NodeMCU, I get garbled text on UART... Have you had any luck with SDKnoWiFi on NodeMCU?

Best regards,
Leandro

cnlohr, 2018-10-16:
I'm glad it helped you out!

I don't think I ever used sdknowifi's build system or bootstrap.  I have used pieces here and there with the nonos sdk, as well as nosdk8266, but, I haven't really played with it as much as I probably should.

I don't know what all setup is required for accessing the RTC, but it might be minimal.  I just don't know :-/

closed 2018-10-16 by cnlohr