cnlohr/nosdk8266 #12: Running at 1.848 GHz but without a UART connection?

0z4, 2021-09-01:
So I was trying to get 462MHz to work playing with the PLL and MAIN_MHZ registers, when I tried to change the led blink rate in main.c, though, for the UART port I was getting only nonsensical garbage (probably because of the strange frequency of the PLL) The led behaved as I had specified, that is, the time between blinks changed. That was when I kept increasing the frequency periodically until I reached 1.848 GHz. I am not very excited by this discovery because surely you had already noticed it while working, but I am thinking, If i implement an I2C communication using GPIO and call_delay_us functions? What do you think about this?

Regards.

cnlohr, 2021-09-14:
WHUT??? @0z4  I have no idea how this was possible.  Can you share more about your various settings?

cnlohr, 2021-09-14:
Please also share any evidence you have surrounding this operation.  I was not able to get the chip to boot at all past the speeds I listed.

I am also incredibly curious what performance you were able to eek out of the I2S port.

0z4, 2021-09-20:
Well, check my fork, i map all possible PLL registers, after disabling the cache and the I2S, when you pass 378MHz the FLASH speeds explode, it can divide any funther. :(

closed 2024-02-06 by cnlohr